petersaj/neuropixels_trajectory_explorer #4: Bregma-lambda pitch?

chris-angeloni, 2023-05-03:
The CCF atlas is rotated 5 degrees nose-down (as noted in the wiki). In the explorer, the brain also seems to be rotated similarly, nose-down slightly. I wanted to check whether this rotation is accounted for during probe angle/insertion values when implanting during the surgery.

Given that my lab levels lambda and bregma during surgery, do I need to account for these 5 degrees when using the angles given by the trajectory explorer?

petersaj, 2023-05-03:
Hi Chris - the angles on the trajectory explorer are relative to leveled bregma-lambda, so all coordinates should be used as-is if you're leveling bregma-lambda during surgery.

To clarify where the tilt comes from: since the CCF was created using only brains and not with reference to the skull, the CCF has an arbitrary tilt relative to the skull. From some recent work, that tilt was empirically estimated to be 5 degrees relative to a level bregma-lambda. The trajectory explorer takes this into account by tilting the CCF and giving all angle measurements relative to the tilted CCF.

chris-angeloni, 2023-05-03:
Great, maybe this would be a useful note to add to the wiki page addressing the atlas!

closed 2023-05-03 by chris-angeloni

petersaj, 2023-05-04:
I tried clarifying on the wiki page, hopefully this makes it clearer? https://github.com/petersaj/neuropixels_trajectory_explorer/wiki/CCF-stereotax-coordinate-conversion