petersaj/neuropixels_trajectory_explorer #9: Error running rat version

cyf203, 2023-07-24:
Hi, 

I am trying to run `neuropixels_trajectory_explorer_rat.m`. I put the rat altas in the same folder as `neuropixels_trajectory_explorer_rat.m` and added to my path. In rat altas folder, it contains 4 files:
```
WHS_SD_rat_atlas_v4.label
WHS_SD_rat_atlas_v4.nii.gz
WHS_SD_brainmask_v1.01.nii.gz
WHS_SD_rat_T2star_v1.01.nii.gz
```

Then, I run the following line in another matlab file:
`neuropixels_trajectory_explorer_rat`

I met this problem: 

```
Error using textscan
Use fopen to generate a valid file identifier.

Error in neuropixels_trajectory_explorer_rat (line 48)
dataArray = textscan(fileID, formatSpec,
'Delimiter', '', 'WhiteSpace', '',
'TextType', 'string', 'HeaderLines'
,startRow-1, 'ReturnOnError', false,
'EndOfLine', '\r\n');
```

I am running this code on window 11 matlab 2023a. 

Thank you!

cyf203, 2023-07-25:
I further tried on 2020b. The same error showed up again.

petersaj, 2023-07-26:
That was from a typo on my end - I've just fixed, do you want to pull the latest version and try again? 

I think you may be the first person to ever use the rat version of this! It's not very up-to-date, but let me know if you find any other issues.

cyf203, 2023-07-26:
Hi, Thank you so much!!! 

We do have a cohort of rat people who are trying neuropixel at Johns Hopkins University. I will spread your fantastic tools with them as well! 

For now, I can run the UI without any obvious problems now.