petersaj/neuropixels_trajectory_explorer #10: Error using rat atlas

AamnaLawrence, 2023-07-26:
Hello,

I was trying to set the entry and end coordinates for looking at the probe placement, however, I got the following error on entering the end coordinates:
**Unrecognized field name "ap_coords".

Error in neuropixels_trajectory_explorer_rat>set_probe_endpoint (line 576)
max_ref_length = norm([max(gui_data.ap_coords);max(gui_data.dv_coords);max(gui_data.ml_coords)]);
 
Error while evaluating UIControl Callback.

Unrecognized field name "ap_coords".
**

I entered the end coordinates as AP: 2.76mm ML: 2mm DV: 4mm Azimuth: 0 and Elevation: 90. Also, I could not change the azimuth angle using the GUI. Any help would be appreciated. Another thing that I wanted to know was if the Paxinos atlas on Sprague Dawley rats can be used. Thank you.

cyf203, 2023-07-27:
Hey, Aamna,

I have the same problem here with using "set end coordinates". 

But I found no problem changing the angle with keyboards. 

As for the atlas, I think this rat atlas should be similar to the Paxinos SD rat atlas. Ideally, it will be good to have a Long Evan rats atlas.

cyf203, 2023-07-27:
I fixed the bug by copying the line define max_ref_length from `set entry coordinates` to the `set end coordinates`. It should be good to run.

petersaj, 2023-07-27:
Made that fix, if you pull the latest repo it should work now.

As far as I know, the Waxholm atlas is the only downloadable annotated 3D rat atlas, and all Paxinos atlases are only available as drawings, but let me know if there are better alternatives.

closed 2023-07-27 by petersaj