petersaj/neuropixels_trajectory_explorer #15: Dropdown menu disappeared

Andrea3v, 2023-08-16:
It seems the dropdown menu (when running within MATLAB) has vanished from both mouse and rat:

[image]

the standalone still has it though. I tried different MATLAB installations but problem persists.
any help to bring it back would be very appreciated!

thank you
Andrea

petersaj, 2023-08-16:
Hi Andrea - it still shows up for me (just mouse version - rat version is lagging behind changes so doesn't have it yet), can you send a screenshot of what yours looks like? Does it appear if you make the window bigger?

Andrea3v, 2023-08-17:
Hi Andy, thank you for the quick reply!
here's a screenshot:
[image]
resizing the window doesn't make it appear, unfortunately.
Just to confirm: the rat version never had it?
I'm mainly interested in rescaling the atlas according to Bregma-lambda distance. I started recording in rats and realized they have a quite significant range of measures (7.5-10mm), is there a workaround I can use to account for that (maybe through the code or something)? 

thank you in advance for any help or suggestion you may have
~Andrea
PS I love this tool!

Andrea3v, 2023-08-17:
you are right! I didn't notice I was running a version from Feb 🤦🏼‍♂️ a conflict was preventing matlab to pull the latest version of the code. indeed the new code has the dropdown! 
thank you for your help and looking forward to the next upgrade!
PS by changing `clim([0 300])` to `caxis([0 300])` on line 137 would help run the code in MATLAB versions before 2022!

petersaj, 2023-08-17:
Great - sure, I switched `clim`s to `caxis`s

closed 2023-08-17 by petersaj