petersaj/neuropixels_trajectory_explorer #17: Connection error interfacing with SpikeGLX

abisi, 2023-11-03:
Hello,

I wanted to try to interface with SpikeGLX, however it did not work. 
I'm wondering if I'm doing something wrong or if it could be a bug.
I'm using the latest pull of NTE.

I followed the instructions on the SGLX webpage:
1. Run a recording (tried both with enabling and disabling recording i.e. saving or not saving)
2. Enabled remoted command server using default IP address, port and timeout. I made sure the command window showed that the software was listening. 
3. On the explorer, connected to SGLX using the same IP address and port

When I clicked connect, I had the following error:
[image]

When I validated the IP adress and port, I had the following error:
[image]

In the SGLX graphical shank map viewer, the "Anatomy colors" box remains empty.
Also, I'm not sure whether the problem comes from the explorer or the SGLX.

Any help would be greatly appreciated! 😄 

Many thanks,
Axel

petersaj, 2023-11-03:
Hi Axel - that's a curious error that it doesn't find a variable in line 2142 that it uses with one line before, it's almost like it got interrupted part-way through execution. 

I can't replicate this, just for sanity: does this replicate on your end every time you try it? And did the SpikeGLX console show any messages about CmdSrv to suggest that stopped working?

abisi, 2023-11-03:
So, not having good reflexes in Matlab, I checked too late that the `nte_` subfolders weren't added the to the path. I just had to do that... 😅 All works nicely. Thanks for the feature!

Axel

closed 2023-11-03 by abisi