petersaj/neuropixels_trajectory_explorer #18: Probe insertion depth not right

zmingmin, 2023-11-21:
Hi,
It seems the probe insertion depth is not right.  Am I missing something? Or is some setting not right?

Thanks a lot!

[image]

petersaj, 2023-11-25:
Can I check: are you working with the latest version of the GUI?

From your picture, my guess is that your probe trajectory is entirely outside the brain, in which case the coordinates do not update. I believe I added an update a while ago that shows a message stating something like "probe trajectory doesn't intersect with the brain", which means that you might be working with slightly old code.

zmingmin, 2023-11-25:
Hi, Andy, thanks for your reply. 
I am using the newest version. 

I cropped two new ones when the probe are completely inside the brain, and the probe depths are still strange (too big).

It seems the data updated in the variable **probe_depths** is not correct. 

Thanks again!

[image]
[image]

petersaj, 2023-11-27:
I wasn't able to replicate your error by going to the coordinates in your screenshot above, I tried all 3 methods of moving the probe (manually, setting entry, setting endpoint) and the depth coordinate looks normal in all cases: 

[image]

Could you tell me the steps you're doing so I could try to get that bug on my end? And is the depth coordinate wrong even when you first open the GUI, or only after you move it?

zmingmin, 2023-11-27:
(When I open the GUI, the depth coordinate is wrong without any movement. )

I found It was the Matlab version issue.  I just updated my 2020 Matlab to 2023 and now it is correct. 

I really like the GUI! Thanks a lot!