petersaj/neuropixels_trajectory_explorer #20: Saving .mat file that includes full list of brain regions across the probe trajectory, not just with bank 0

codylslater, 2024-01-27:
I have been doing implants using the MPM micromanipulator system, while streaming brain regions to SpikeGLX. I perform an acute recording at the time of implant, but actually head fix the probe and longitudinally record for several weeks. Right now when I save the probe positions to load into the explorer later - it only saves region positions for the bottom bank of the probe. I am actually inserting >3.84 mm, and later configure the .imro file for custom recording up and down the probe. I would like to be able to load the positions from the original acute recording at implant, and it would be easier if these were in the .mat file output by Neuropixels Trajectory Explorer, perhaps with a mask for whether or not the region is aligned with Bank 0. Is this something that would be useful to add into the saved .mat file for positions or would you recommend I proceed recovering the regions from meta data saved in SpikeGLX?

Thanks for the great tool!

petersaj, 2024-01-27:
Hi Cody - I'll have to change the way NTE saves in the (hopefully) near future to work with 4-shank probes, but in the meantime: can I check how you'll be using the data? If SpikeGLX saves the regional metadata (I haven't used it myself) then that seems like a good option for quick analyses during ongoing experiments, and ideally the planner/recording anatomy would be replaced with histology once you have it.

petersaj, 2024-02-01:
Makes sense, that's a better way to save things. I've just made these changes: 

- it now saves all areas along the probe trajectory
- the distance has been changed from "probe_depth" (which was essentially arbitrary being relative to bank 0) to "probe_tip_distance", where negative numbers are away from the circuit board, and positive numbers are towards the circuit board (e.g. top of the probe)

Do you want to try this out and see if it hits your needs?

codylslater, 2024-02-07:
Sorry for the delayed reply. The new saving method is great - gives exactly what we need. Thank you!

closed 2024-02-07 by codylslater