petersaj/neuropixels_trajectory_explorer #21: Offline use?

GuilleSGCUPO, 2024-02-12:
Hello, thank you very much for providing such an useful and impressive tool for Neuropixels recordings. I have a question regarding its capabilities: Can I use past Neuropixels sessions with all the data regarding depth, angles and stereotaxic regions to obtain results, or does it only work with online recordings? Thank you once again for your assistance :)

petersaj, 2024-02-12:
You can save and load probe positions in the Save/Load menu (right-most menu bar option), does that feature do the things you're looking for?

GuilleSGCUPO, 2024-02-12:
I apologize if I didn't explain myself clearly. I'm interested in utilizing past Neuropixels recordings, which are saved in SpikeGLX, to verify the placement of my channels. While exploring the Save/Load menu, I noticed it requires a .mat file containing position data. Unfortunately, I don't have this information stored in a MATLAB archive; it's documented on paper instead. Is there a method to utilize this information?

petersaj, 2024-02-16:
SpikeGLX doesn't save estimated probe location in 3D, so you can't get back the position from that - the NTE does though, so if you're linking your manipulator to the NTE, you can save the probe locations after insertion. 

Apologies I'm still not clear on exactly what you're trying to do: what data do you have, and what analysis do you want to do?

GuilleSGCUPO, 2024-02-20:
Sorry, maybe my explanation was a bit poor. I have data from old SpikeGLX recordings, with neuronal spikes data through mostly all channels. Also, I have the stereotaxic coordinates for the region I inserted the NPs. So, my question was if there is a possibility for NTE to work with these old (offline) recordings, for I have seen, in the github, that it works in online recordings, during the recording and I wonder if there is a chance to carry out the analysis after the recording ends.
Thanks again for your attention and I hope I was clearer now :)

petersaj, 2024-02-20:
I see - there's not a way for the NTE to load positions via SpikeGLX, although if you have the stereotaxic coordinates, you could always set the probe up at those coordinates in the NTE (and save, if that information is useful). Does that help?

GuilleSGCUPO, 2024-02-22:
Thank you for the clarification! I'll explore using the NTE offline and set up the probe based on the stereotaxic coordinates I have. Saving that information could indeed prove useful.
I appreciate your attention to this matter! :)

closed 2024-02-22 by petersaj