petersaj/neuropixels_trajectory_explorer #22: Startup failure with standalone application when CCF atlas directory changed

wangxuanyu0419, 2024-05-27:
Hi Peter!
I had an issue with the undeployed version of the Neuropixels trajectory explorer. I moved the CCF files to another directory because I ran out of disk storage. However, the software does not start with a relocation request but terminates with the following error:
```
File not found: xxxxx\template_volume_10um.npy
Error in => neuropixels_trajectory_explorer.m at line 80
```
I tried to reinstall the software, or change the `nte_paths.mat` under `./nte_deployed`. Neither worked.
I wonder where is the file location cashed and can I remove this history?

petersaj, 2024-05-27:
Hi Xuanyu - just checking: do you mean the deployed version (e.g. without Matlab, the undeployed version run through Matlab just needs the folder with the CCF to be added to the Matlab path)? 

I've uploaded a fix where it checks for the files and asks for the path again if it can't find them - do you want to update and see if the problem is fixed for you?

petersaj, 2024-05-27:
Yes, I fixed the `.exe` file (that's the deployed version) for for the issue you mentioned, if you wanted to update the code and try it

wangxuanyu0419, 2024-05-27:
Sadly, I still got the same error. I uninstalled it via the Windows control panel and pulled it locally, reinstalled it with the `.exe`. The error message was still there.

wangxuanyu0419, 2024-05-27:
The undeployed version was working fine.

petersaj, 2024-05-27:
Oops apologies, I forgot to copy the updated installer into the main page - I've done that now, do you want to try again?

wangxuanyu0419, 2024-05-28:
Yes, all set! Thank you!

petersaj, 2024-05-28:
I think I found the issue - could you pull the latest version and try again?

closed 2024-08-06 by petersaj