petersaj/neuropixels_trajectory_explorer #23: SpikeGLX connection and Saving Probe Areas Error

NecLabResources, 2024-05-30:
The latest version of NTE seems to run into errors when interfacing with the MPM pathfinder and SpikeGLX. 
1. When I use MPM pathfinder to drive the probe to my desired location, I wish to save this probe-areas.mat but get the following error:
"Index in position 1 exceeds array bounds. Index must not exceed 1.

Error in neuropixels_trajectory_explorer>save_probe_positions (line 1727)
    'rotation_angle',num2cell(probe_angles_cat(3,:)));
 
Error while evaluating Menu Callback."

2.  Additionally, when I try to establish a connection to SGLX, I get the following error:
"Unrecognized function or variable 'probe_depths'.

Error in neuropixels_trajectory_explorer>update_probe_areas_coordinates (line 915)
        probe_depths, ...

Error in neuropixels_trajectory_explorer>connect_spikeglx (line 2379)
update_probe_areas_coordinates(probe_atlas_gui);
 
Error while evaluating Menu Callback."

For the SGLX issue, I db-stepped to the " send_recording_areas" function and saw that the following variables are no longer part of the workspace (rather, it seems they have been replaced by similarly named variables, slightly different enough to break this function call?)

Any help clarifying/solving these two issues would be greatly appreciated!

[image]

petersaj, 2024-05-30:
Hopefully both issues should now be fixed, if you want to pull the latest code and try again? 

(1) Looks like it was from an issue that was fixed previously, so you might not've been using the latest code. 
(2) I see the issue, I think fixed on my end but let me know if there are issues

closed 2024-08-06 by petersaj

NecLabResources, 2024-08-09:
Thanks for responding with a fix so quickly--and, apologies for the delay in getting back to you.
Yes, I see now after pulling the latest code changes it looks like these no longer pose any issues. 

Thank you!