petersaj/neuropixels_trajectory_explorer #25: MPM Pathfinder unrecognized 'ShankCount' Property on connection

NecLabResources, 2024-08-09:
Accessing the MPM Pathfinder through NTE poses this error on connection (via localhost, port 8080)

"Error using neuropixels_trajectory_explorer>connect_newscale
No appropriate method, property, or field 'ShankCount' for class
'NstMpmClientAccess.NstMpmSvrProbe'.
 
Error while evaluating Menu Callback."

This is after I've started the neuropixels trajectory explorer, added a probe, and driven it to a select location.

Any help resolving or clarifying the cause of the issue would be greatly appreciated!

petersaj, 2024-08-12:
Sounds like you may have an old version of Pathfinder from New Scale, that was introduced in v2.8.8.

You can download the latest from this page under "Software": 
https://www.newscaletech.com/download-mpm-cad-files-software-user-guides/#pathway

You can check whether you have the right version of Pathfinder by clicking on a probe letter (e.g. A) to get the probe properties box, and the "Probe" tab should have a "Multi-Shank Probe" section at the bottom: 

[image]

NecLabResources, 2024-08-25:
Yes, thank you! We do now have the latest version of Pathfinder installed and I can now connect to both pathfinder and SGLX without issue. 

Though the probe positions can be saved using NTE alone (prior to any peripheral software connections), NTE now presents the following error when trying to save a probe positions file after a connection to MPM has been made (and even after MPM has been disconnected).

"Index in position 1 exceeds array bounds. Index must not exceed 1.

Error in neuropixels_trajectory_explorer>save_probe_positions (line 1701)
    'rotation_angle',num2cell(probe_angles_cat(3,:)));
 
Error while evaluating Menu Callback."

Wondering if this is some key trajectory parameter related to the NTE backend that MPM overwrites. Any help is, again, much appreciated.

NecLabResources, 2024-08-28:
Update and for reference: This appears only to present an error when a probe has already been instantiated from NTE prior to an MPM connection. The probe map saving was successful when connecting the MPM without a prior probe instance.

closed 2024-08-29 by petersaj