petersaj/neuropixels_trajectory_explorer #27: is that possible to change the neuropixiel probe to surgery forceps

zclecle2, 2024-12-02:
Thank you for your work!
I am wondering if it is possible to change the neuropixiel probe to surgery forceps (like the one shown in the following). 
And this might be helpful for designing other surgery experiments.
Thank you!
[image]

petersaj, 2024-12-02:
Hi - there aren't plans to add anything that doesn't go into the brain, I might refer you to Pinpoint (https://data.virtualbrainlab.org/Pinpoint/) which renders some kinds of peripheral tools and devices.

closed 2024-12-02 by petersaj

zclecle2, 2024-12-02:
Actually, the forcepts will go into the brain to create an injury model. We'd like to show others how to perform such surgery.

petersaj, 2024-12-02:
I see - I don't think I can implement that change in the explorer, but hopefully the tip location of the "probe" can still give you the type of targeting info you need