petersaj/neuropixels_trajectory_explorer #28: Displayed slice does not change with perspective for NP2.0

mik-schutte, 2025-01-24:
Hi,
When planning a trajectory for NP2.0 probes I run into the following bug. The displayed slice (either anatomical or annotaded) does not move based on the perspective but in stead remains parallel to the shank (see image).
This problem does not occur when I plan a trajectory for 1.0 probe and is also not resolved with a fresh install of the trajectory explorer.

MATLAB version = R2024b
OS: Windows 11 version 24H2

Please let me know if you are in need of additional information.

[image]

petersaj, 2025-01-25:
Hi - this is meant to be a feature rather than a bug: the slice locks to the plane of the shanks, which is undefined for the 1-shank but defined for the 4-shank. That way, you're able to see exactly where each shank is in the slice, which would be difficult if you had to manually rotate it to match yourself. 

Is there a reason you'd prefer it to rotate with the 4-shank? If you want to see what's in the neighborhood while you're planning a trajectory, you could alternatively try plotting nearby 3d structures or rotating the probe?

mik-schutte, 2025-01-27:
I see your reasoning behind the feature. My goal was to use both the known insertion region and the probe dye tracts (from coronal sections) for increasing certainty for determining the probe location during experiments.

However, I have found a way to make it work now. Thanks for your reply.

closed 2025-01-27 by mik-schutte